alexqzd/Marlin #7: Bug found when trying to print object with jpg image embedded

n9jcv, 2021-12-18:
Hello Alex

### Description

Here is a bug I have found, I do not know if it is in the firmware itself, or a problem with the python script.  It is an intermittent bug, that does not happen all the time, and I do not yet know exactly when or why, but it happens 50% of the time.

I use Prusa Slicer 2.3.0, Windows 10.  Using an Aquila X2, printing from SD card not connected to anything.
I click Print, then I scroll thru the list and find the object I want to print.  It does display the small image of the item and it is correct.  I click the object, the screen thinks for a few seconds, then I see a blue progress bar, and when the progress bar finishes, it goes back to the home menu.  It never starts to print and it does not heat or do anything else.

Sometimes, I just reslice the object and then it works.  Othertimes, I can not get it to work, so I end up saving to the card without the image, just so I can print it. 

When I slice, with the image python script, it slices fine.  Then when I export, I see a black window appear for a second, to run the python script and then the black window goes away, so I cant tell if there are any errors in the python script.  I tried to look for logs but could not find any.

I would be glad to try and debug with you.

Thanks for all your efforts and this amazing contribution.
n9jcv

tititopher68-dev, 2021-12-30:
@n9jcv 
I think this is from encoding the image in PrusaSlicer and therefore the script, from the Python version.
Modify the python script by replacing at the beginning of the file, python by python3
And take a test again.  Post-processings in Prusa require python3.

stas2z, 2022-01-01:
Same with cura

tititopher68-dev, 2022-01-01:
@stas2z @n9jcv 
I confirm for both Prusa and Cura.  It is ultimately not the script, but in the firmware.  I have done a lot of testing and it happens when the weight of the kb image exceeds 5.5 kb.  When the buffer limit is reached, the preview is not displayed and often, as this generates a processing error, it crashes all the firmware and therefore reboots.  Also, I will look to optimize and find the right compromise between image size, compression rate and buffer size.  If Alex could intervene, it would be even better.