alexqzd/Marlin #8: Thermal Runaway problems

ceorvis, 2021-12-27:
Using any firmware versions for BLTouch auto-level, except the 9x9 grid, triggers erroneous Thermal Runaway warnings during leveling.  There is no thermal problem, temperatures are normal.

stijnb1234, 2021-12-27:
Which version of the printer do you have, and which version of the firmware are you on?

ceorvis, 2021-12-27:
I have a G32 Aquila, running v1.3.5 BLTouch-9x9-G32.bin
This version of 9x9 is stable for me most of the time.  All other G32 versions of 1.3.5 (5x5, 4x4 and 3x3) all trigger the Thermal Runaway error during pre-print leveling, and all G32 versions of v1.3.4.1 including 9x9 have the same issue.

Roboineer, 2022-01-04:
i too am having this problem, has anyone found a solution to this? i have a N32 version

ceorvis, 2022-01-04:
My "solution" is to manually set the Extruder temp to 0 before leveling starts.  Whatever the bug is, it doesn't occur if the hot end is not trying to heat up while probing.

For me the bug occurs any time the touch sensor is probing.  This means: Setting the auto-level mesh outside of a print, Pre-leveling before every print, even during Homing.  Any time the probe touches the bed surface if the hot end is set anywhere above 0, it absolutely freaks out and demands to be restarted.

Roboineer, 2022-01-04:
I get the error when printing as well, it doesn't get the error in idle soo idk why it's doing this

ceorvis, 2022-01-04:
I have noticed that it only happens when the extruder is heating while the touch sensor is probing something, and never when only one of those conditions is met.  

I've also noticed that while it is probing/leveling, if I attempt to interact with a menu item (such as setting a temperature or changing the Z offset) that my Aquila will not respond to input while the probe light is blue.  I wonder if these two things are related somehow.  If the firmware is actively monitoring temperature (for a value above X for Y seconds) and loses control/input during a probe cycle while the temperature is right at the set point or maybe one degree above then is the 3-4 seconds it is "without input" enough to trigger a thermal warning when really the temperature is responding as commanded?

ripptech, 2022-01-06:
throw my hat in here.  VL Aquila G32.  updated to "noprobe" from stock and it worked great.  installed a 3d-touch, the G32 9x9 BL firmware, and immediately started getting the kill errors with "heating failed"...  just switched back to the noprobe firmware and its printing away happily again.  As above I would get these halts randomly when trying to auto-level, or soon after starting a print.

ceorvis, 2022-01-06:
Did you happen to try setting the extruded temperature to 0 manually before printing or leveling?  This worked for me, if it works for everyone it could be a simple thing for the firmware developer to implement rather than troubleshooting the actual root issue which would give us a faster fix.

gminor7, 2022-01-10:
I'm having the same issue with Aquila x2 N32 bltouch 3x3 (BLTouch-3x3-N32.bin).  I leveled with the heat off and saved the mesh to the eeprom. I had to remove the g29 code from Cura and replace it with M420 S1, otherwise the temp goes all over the place when the probing starts and it gives off a thermal runaway error.

Roboineer, 2022-01-17:
does anyone might know what could be causing it in the source code or whats being triggered? might try going into the source code and tweaking it soo it doesnt cause these problems

Geo011, 2022-01-21:
I'm having the exact same issue. Installed the BLTouch-3x3-N32.bin today, and getting Thermal Runaway errors. When installing the non-bltouch firmware the printer works fine. Using the M420 S1 code also works. But i would like to probe my bed everytime i start printing.

iowacountryboy, 2022-01-29:
Same issue here on my Aquila N32.  I've tried the BLTouch-3x3, 5x5 and the HS versions with the same issue.  I have not tried the 9x9 yet but I will today.  I also found as previous posters mentioned, if I set the hot end temp to 0 before creating a new mesh the issue does not occur, otherwise the thermal runaway issue happens every single time if the hot end temp is anything other than 0

Speeedy0815, 2022-04-08:
Me too,

Firmware: Aquila N32. BLTouch-3x3

The workarround (heating after Probe) is working for me.

For me it seems, that during the measurment  of the pad the PID is not wokring. Somtime it is heating up a little bit. (no Problem)

and sometimes it's cooling down, and than the underheatprotection is causing an error.