alexqzd/Marlin #9: N32 Aquila - Firmware won't install

peliopoulos, 2021-12-29:
### Description
Firmware won't install on new Voxelab Aquila N32

### Steps to Reproduce
1. Formatted microSD card with 4096K allocation unit size, FAT32
2. Downloaded  Default-NoProbe-N32.bin and copied to root folder of card
3. Inserted card into brand new Voxelab Aquila N32
4. Turned printer off, waited 10 seconds and then on
5. Printer booted to factory firmware immediately

**Expected behavior:** 
Alex's Marlin firmware installed on printer

**Actual behavior:** 
Nothing happened, no error message, regular firmware runs

#### Additional Information
I tried this with two different cards
I tried re-downloading .bin file
I tried it with the 3X3 Mesh file

Please help!!

Mik-S-UMS, 2021-12-30:
Place the .bin file inside a folder called firmware

peliopoulos, 2021-12-30:
thanks @Mik-S-UMS !  that solved that problem, but now I have new ones that will be a new issue (once I do some reddit searches)

closed 2021-12-30 by peliopoulos