alexqzd/Marlin #10: Can't see Gcode Previews

rainy1505, 2022-01-02:
Hello,

Cura 4.12.1
Flashed my Voxelab Aquila G32 with Default-NoProbe-G32.bin and Alex's Firmware v1.3.5.1 Beta 4.
Control -> Visual -> Show file Thumbnails is checked

When I slice something I can see that my .gcode files now includes base64 image:
[image]

Tried to restart printer. Enable/disable show file thumbnails. But still I can't get it to work.

tititopher68-dev, 2022-01-03:
@GitUser999A 
Tu as pris le fichier gcode chez mriscoc ? C'est donc normal que cela ne fonctionne pas !

To include PrusaSlicer thumbnails:

Make a copy of the the printer profile, and go to menu>help>Show configuration folder.
Locate the printers subfolder and edit the .ini file of your printer.
Around the bottom (line 69), there is a line that says "`thumbnails =`" change it to "`thumbnails = 50x50,217x217`".
Then add PrusaSlicer_JPEG_Preview.py to the to the post-processing scripts section of print settings : 

1- Installation of Python (if not already installed) : `https://www.python.org/downloads/release/python-399/`
2- Copy the script '`PrusaSlicer_JPEG_Preview.py`' in a folder ie :  
`C:\Program Files\Prusa3D\Scripts_Post-processings\`
3- Open PrusaSlicer and go to Print Settings Tab, Post-processing scripts section and enter exactly in the input field `"C:\Program Files\Python39\python.exe" "C:\Program Files\Prusa3D\Scripts_Post-processings\PrusaSlicer_JPEG_Preview.py";`

[image]

4- Next go to Printer Settings tab, Firmware section and check if you have `50x50, 217x217`. If not enter this parameter in the input field.

[image]

5- Don't forget to save your new profil.

To include Cura thumbnails : 

1- Go to the Help menu on Cura and select Show Configuration Folder in Windows you will get an explorer opened in the path inside of your user profile folder: `C:\Users\'user name'\AppData\Roaming\cura\`
2- Close Cura and save the script 'Cura_JPEG_Preview.py' in the scripts folder.
3- Open Cura, go to menu `Extensions -> Post Processing -> Modify G-Code` and press Add a script, then select the script  `Create JPEG Thumbnail` from the post processing plugin popup.
4- Select for Width and Height values : 50 and validate.
5- Again, go to menu `Extensions -> Post Processing -> Modify G-Code` and press Add a script, then select the script  `Create JPEG Thumbnail` from the post processing plugin popup.
6- And select for Width and Height values : 217 and validate.

Pictures from prusaslicer :  (from @alexqzd) : 
[image]
[image]

Nota : 
Malheureusement, si je ne sais pas si @alexqzd continue ou pas les fix concernant le Gcode Preview, car selon la taille du Preview, cela peut s'afficher ou pas

GitUser999A, 2022-01-04:
Can someone give me file which includes thumbnail and works for you, so I can do some testing?

tititopher68-dev, 2022-01-04:
@GitUser999A 
This is good but don't forget that if you have one or more Gcodes files with images weighing more than 5.5 kb, the preview part will not be displayed. The image is encoded in base 64. the weight of the image does not depend mainly on its 217x217 size, but depends greatly on the dimensions of the object to be displayed.
On the Jyers site, I modified the code to decrease the size to 180x180 and increasing the compression by 5%, which does not harm the display. And of course in the firmware code I also made the necessary changes. So the same Gcode encoded in 217x217 is not displayed, now is displayed in 180x180 ... Even with a very large object, I manage not to exceed the size of the maximum buffer.

stijnb1234, 2022-01-04:
@GitUser999A If resolved, please close the issue. ;)

closed 2022-01-04 by GitUser999A