alexqzd/Marlin #12: failure to load Gcode when Print is larger

Dpapesca, 2022-01-20:
### Description

weird issue that seems to depend on file size. failure to load the Gcode and start the print when I try and select a larger or more complicated file. when I select something like a benchy or XYZ cube the printer operates as expected. I was running this newest beta version when this issue occurred. I revered back the the version right before this one and it worked properly which is why I am inputting the issue request. I have included the file I was attempting to print below in step one.

### Steps to Reproduce

1. I downloaded this torture test to well test my printer.  https://www.thingiverse.com/thing:3561875
2. brought it into cura and sliced it according to the settings needed buy the filament ( esun pla+)
3. after saving it to my micro SD I loaded it into the printer 
4. upon selecting it under the print menu it actually shows a loading bar ( which never happens) after the bar completes it just brings me back to the main menu. 

**Expected behavior:** I expected it to ask me to confirm the print I selected then print 

**Actual behavior:**  a loading bar appears then it sends me back to the main menu. 

#### Additional Information

I am running the BL touch 9x9 G32 HS pre configured file on a voxelab aquila 

I apologize if I am not using the correct forum for this. I am new to GitHub and genuinely believe I found a bug in this latest beta version and wanted to bring it to your attention.

n9jcv, 2022-01-21:
You failed to mention what version of alex fw you are using.  I have a feeling you are using his gcode preview version.  If so, this is a known bug.  If not report back the exact version number and file name of the bin file you loaded.

alexqzd, 2022-03-12:
Try generating the preview with this updated plugin https://github.com/alexqzd/Marlin/blob/5d4286e1a80797d4c11041149063a5e1a53b7c11/Slicer%20post-processing%20scripts/Cura_JPEG_Preview.py