alexqzd/Marlin #13: Cannot install (again)

Tousakusha, 2022-01-26:
### Description

I cannot install the Manual-Mesh N32 Firmware !again!.

### Steps to Reproduce

1. I Updated from Stock-Firmware to Manual-Mesh N32
2. Later i Bought a 3D Touch, upgraded to BL-Touch Firmware N32
3. 3D Touch Broke and i wanted to go back to Manual Mesh N32

**Expected behavior:** Now i expected the Printer to have the Manual-Mesh Firmware.

**Actual behavior:** But it will not install

#### Additional Information

I could install the Stock-Firmware without problem. But when i try to upgrade now it looks like this:
https://ibb.co/4gndRJ4

jrb4562, 2022-02-11:
@Tousakusha did you find a solution to this? I am seeing the same error although I cannot even get the original firmware to install and get back into the menu now

classicrocker883, 2022-02-16:
@Tousakusha try renaming the firmware file. put in a folder called firmware.
if that doesnt work put in root folder, which is directly on the drive not in any folder.

jordanmarchetto, 2022-03-08:
I'm having a similar issue.

I had manual mesh g32 firmware installed, I bought a cr touch, I tried to install the bltouch firmware, and it doesn't seem to be working.  I still see manual mesh settings, I don't see any bltouch settings.