alexqzd/Marlin #14: Black Screen After install N32 v1.3.5

jrb4562, 2022-02-11:
I installed the BLTouch-3x3-N32 version and all seemed to go well and I got the progress bar and checkbox however the system never booted into the menu again and just stayed on this screen
[image]

After removing the SD card and rebooting I am now stuck on a black screen with backlight. I have attempted to revert to OOB firmware with the same results. Any ideas?

classicrocker883, 2022-02-16:
https://github.com/classicrocker883/Marlin/releases/tag/v1.3.5

try my fork.
I know there isn't BL touch 3x3 but since u have one try the Ubl7x7. it's faster and overall better. you get more points to probe the bed for a more accurate level

classicrocker883, 2022-02-27:
> Works fine for me. Only 'problem' I have with it, is a white bar at top, and the changes in font color (like aqua / orange).

I had forgotten to change back to default colors when I compiled firmware for my own. 
but you can go into advanced settings and change it to whatever you'd like - sorry for the inconvenience