alexqzd/Marlin #22: NEED HELP

davidneighbors, 2022-03-16:
I have a Voxelab Aquila that's brand new got a CRtouch for it downloaded the 10x10 firmware for it for my G32 chip. Everything went great until I got to bed leveling. When I do manual leveling with probe it's telling me my corners need to be raised I raise them and get them very close but the center of the bed is very low but when I create a mesh it says the center is dead nuts and all my corners are really high!? I have changed out the bed to a PEI sheet I've changed out the springs to silicone spacers I've tried different ones of Alex's firmware all with the same issue consisting even tried a buddies probe that we KNOW works and getting the exact same thing. Any help?? THANKS IN ADVANCE!

junland, 2022-03-21:
Did you set / calibrate the Z-offset? Also I would take your question to [r/VoxelabAquila](https://www.reddit.com/r/VoxelabAquila/) on Reddit as you may get better answers.

davidneighbors, 2022-03-22:
Yessir I did. I even got ahold of Voxelab customer support and now have it's one way back for my money back unfortunately I believe the motherboard is bad. I will be ordering a creality ender 3 v2 here shortly.

junland, 2022-03-24:
> Yessir I did. I even got ahold of Voxelab customer support and now have it on its way for my money back unfortunately I believe the motherboard is bad. I will be ordering a creality ender 3 v2 here shortly.

Gotcha. I would go ahead and close this issue as to not take up space on the issue list.

classicrocker883, 2022-05-28:
https://github.com/classicrocker883/Marlin/releases/tag/v2.0.1

try my UBL 7x7 instead. it's better than the 10x10, and probes more of the bed.

to get the best level, you have to first get your Z offset corrected. I find not using anything between the nozzle or bed to work bed, the goal is to have the printer know where the nozzle touches the bed. so when it is at Z 0, the nozzle jjjjjust barely touches the bed surface. in the Z offset menu, that's where u do this.

by adjusting your Z offset so the nozzle touches the bed, that is your Z offset. now then level your bed. you'll get it dead nuts close to 0.00 as possible. what I do is go to the center of the bed and get the auto level reading for that, then I go to each corner and copy that reading with the knobs. doesn't matter if it's auto or manual, as long as it's all the same touching the bed as it is the center for edges you're good.

closed 2022-08-28 by davidneighbors