alexqzd/Marlin #30: new Marlin

314ru, 2022-08-20:
A new Marlin has come out. Are there plans to release a new firmware version with support for LA and S-curve?

classicrocker883, 2022-10-06:
@314ru I'm actually working on that. https://github.com/classicrocker883/MriscocProUI

I've been able to get it to work, but there may be some issues with clones of the BL touch.

NurikNar, 2023-02-13:
Hello. I would like to ask you for help. If anyone else can help I would be very happy. I want to say in advance that you have made a great contribution to custom 3D printing, and I am very grateful to you for this.
I state the essence of the problem.
I have an Aquila X2 3D printer with H32 motherboard.
Yes, I used your ready firmware .bin file. But there is a limitation on the maximum heating of the heating nozzle by 260 degrees, I would like to increase this value to 300, since my Extruder allows me to heat up to this temperature. Can someone send the finished firmware with this setting ???
I wanted to make a firmware for this printer myself, but I can’t assemble the firmware (Marlin) for this board, since there is no such option in the list of motherboards (chips). There are on G 32 and N 32. If you compile for these boards or some suitable ones, then a black screen comes out after the printer is flashed.
I'm sorry, maybe I'm asking stupid questions, but I started to study this topic relatively recently. I followed all instructions. Used all the latest updates.
Oh yes and more. I flashed your Marlin configuration with my Aquila s 2 settings and everything worked for me.
Thank you for that. I hope someone can help me with my problem with Aquila X2. Thank you all in advance.
nuriknar2016@gmail.com