alexqzd/Marlin #33: Feature Request: Send the finished print to the front of the printer

rbreeden53, 2022-08-27:
### Description

When a print is complete, send the print bed to the front of the print. This puts the print in a position where it is ready to be taken off of the bed and prevents the feedback to the card of manually pulling the bed forward.

This could be set as an option for automatically moving the bed when a print is complete, or as a menu option to move the bed to y=max

### Steps to Reproduce

Not a bug

**Expected behavior:** [What you expect to happen]

When a print is completed, the print bed moves to the front of the printer, instead of the back of the printer.

**Actual behavior:** [What actually happens]

When a print is completed, the print bed moved to the back of the print and needs to be pulled forward in order to remove the print.

#### Additional Information

* Include a ZIP file containing your `Configuration.h` and `Configuration_adv.h` files.
* Provide pictures or links to videos that clearly demonstrate the issue.
* See [How Can I Contribute](#how-can-i-contribute) for additional guidelines.

Mik-S-UMS, 2022-08-27:
Just add     `G1 X0 Y220 F2000.0 	;Move bed out for part removal` to your end G-code

rbreeden53, 2022-08-27:
Yes, however that requires the editing of the gcode when the gcode is given to you, and you can't modify the configuration (compliance issue). 

Thank you for the actual gcode. I was going to be looking for it to add to the post steps of the slicer, and to document for a Change Request.

classicrocker883, 2022-09-08:
>  I also just found out that the slicer being used does not have a place to add the post processing commands.

@rbreeden53 just paste it into the gcode file using a text editor after it's been sliced, and before u load to printer