alexqzd/Marlin #35: Pause to Print return to Z-axis too high

kyledurieux, 2022-09-24:
Using a Voxelab Aquila S2, I set the pause to print to stop so that I could change the filament. (This issue also occurred with the change filament plugin as well). Machine paused at 14mm and then moved to park position at 15mm on the z axis. 
Fine. 
Then once the pause was complete the machine returned to print higher than the last printing position, about 1mm higher. 

I read through a few blogs on the issue, here is one: (https://github.com/Ultimaker/Cura/issues/5924). I am hoping that the issue can be rectified soon by Cura? Or, someone with the technical know how to fix the error. Right now I am manually changing the g-code to fix the issue.

classicrocker883, 2022-10-06:
how about redo last layer? there's an option for that, no?

kyledurieux, 2022-10-10:
There is an option there for that yes but why would I choose that option…? I admit I did not try that option yet because it didn’t seem relevant to use that option, but I will try it.