alexqzd/Marlin #37: Voxelab Aries support (PN32 based)

nerdosity, 2022-10-25:
### Description

Hello, Voxelab released Aries almost an year ago.
It's based on the PN32 AFAIK.
I have one, it's easily enclosable and the frame is quite good.

https://3dprintbeginner.com/voxelab-aries-review/

I can provide some support, I'm a developer myself (web one, thought I had my dose of C during university) and I managed to compile marlin for my modded Chiron in a couple of days.

How can I help to develop a working version for our printer?

I can solder, too, and I'm not afraid to solder a USB connector on the motherboard if we think it's just unpopulated but still working.

Please let me know how can I help

classicrocker883, 2022-12-29:
I have this printer too. and it is a decent machine.

unfortunately it isn't easy, or maybe it's not possible - to have a different firmware.

but what is PN32? how does it relate to the Aries?

so in order to have a different firmware, besides getting a different motherboard and LCD, you would need to connect to the board.

I believe it's possible to connect using an ST link. but there are several components for the USB missing if you wanted to do it that way. but I have seen a hack where someone soldered wires and connected TX and RX (and ground and or power) right to the chip. this may be the simplest way to interface for changing the firmware. however, if there was only a way so that it's compatible with the LCD. if you can somehow figure out the lcd, the code and compatibility like in VS code, then you can edit the menu, or make a new one.

so there would be a lot of work mostly finding the source code, if voxelab can release it, that would help enormously.

NurikNar, 2023-02-13:
Hello. I would like to ask you for help. If anyone else can help I would be very happy. I want to say in advance that you have made a great contribution to custom 3D printing, and I am very grateful to you for this.
I state the essence of the problem.
I have an Aquila X2 3D printer with H32 motherboard.
Yes, I used your ready firmware .bin file. But there is a limitation on the maximum heating of the heating nozzle by 260 degrees, I would like to increase this value to 300, since my Extruder allows me to heat up to this temperature. Can someone send the finished firmware with this setting ???
I wanted to make a firmware for this printer myself, but I can’t assemble the firmware (Marlin) for this board, since there is no such option in the list of motherboards (chips). There are on G 32 and N 32. If you compile for these boards or some suitable ones, then a black screen comes out after the printer is flashed.
I'm sorry, maybe I'm asking stupid questions, but I started to study this topic relatively recently. I followed all instructions. Used all the latest updates.
Oh yes and more. I flashed your Marlin configuration with my Aquila s 2 settings and everything worked for me.
Thank you for that. I hope someone can help me with my problem with Aquila X2. Thank you all in advance.
nuriknar2016@gmail.com