alexqzd/Marlin #40: Is G29 P3 included in 'Create New Mesh'

vica153, 2022-12-09:
When I use the LCD screen to 'Create New Mesh', does it run G29 P1 then G29 P3 or just P1?

My printer start code includes G29 L0, but do I need to add G29 P3 after that?  I would guess it has already been done. I just wanted to check.

classicrocker883, 2022-12-27:
I think it is G29 P1, because it automatically estimates for unprobed spots. 10x10 really only probes half the bed because of the limitation with the probe. as u can see when it's all the way to the right, there is still part of the bed it cannot read. so either u can manually adjust for it, or accept the estimate it gives, which is pretty accurate.

I have firmware on my repo that uses 7x7 UBL. this actually probes 42/49, leaving the last row, but it's better than almost half the bed with 10x10.

as for start code. use M420 S1 L0, if u want add Z10 for fade height.

vica153, 2022-12-27:
Looking at the mesh viewer it looks like it has extended the mesh to the area it could not probe. This makes me think the 'Create new mesh' runs P1 and P3, so start code should only need L0.

My 10x10 only hits 60 of the the 100 points.  I have thought about remaking the shroud to fit the probe closer to the nozzle and maybe get to 70-80 of the points, but with a reasonably flat build surface 60/100 is more than enough.