alexqzd/Marlin #41: Live Z-Offset not working

inspectionsbybob, 2023-02-08:
### Description

When Trying to adjust the Z-Offset, clicking the "live" option does nothing.  Rotating the Knob does nothing until you click, at which time it moves to the indicated position.

Revision:  In double/triple checking what I was writing, moving between Microsteps and the Knob to make gross changes, it is inconsistent.   SOMETIMES it re-homes and then goes into the adjust mode, other times it just sits.  If you are using the micro-steps, the Knob can't be used for gross changes, and vice-versa.  Very confusing/frustrating.

### Steps to Reproduce

1. Home Z
2. Set Z offset to something safe (positive value)
3. Set Z to 0.0
4. Go to Set Z-Offset
5. Click Live Set
6. try to use BOTH the Knob and the Micro-steps.  I Tried to use the Knob to make Gross changes as I knew the Offset was somewhere around -1.55 so I used the Knob to go to -1.45 and then wanted to use the Microsteps to sneak up on the exact number slowly.
 
    Somewhere in there it will "reset" and re-home Z, and then which ever you use FIRST will be the only one you can use.

If "Live" is set I expected to be able to use both the Knob and Micro-steps back and forth without the printer "randomly" deciding to do a re-home.

**Actual behavior:** [What actually happens]

#### Additional Information

I am using your pre-compiled G32-BLTouch-9x9 so I don't have a Config or adv-config.

NurikNar, 2023-02-13:
Hello. I would like to ask you for help. If anyone else can help I would be very happy. I want to say in advance that you have made a great contribution to custom 3D printing, and I am very grateful to you for this.
I state the essence of the problem.
I have an Aquila X2 3D printer with H32 motherboard.
Yes, I used your ready firmware .bin file. But there is a limitation on the maximum heating of the heating nozzle by 260 degrees, I would like to increase this value to 300, since my Extruder allows me to heat up to this temperature. Can someone send the finished firmware with this setting ???
I wanted to make a firmware for this printer myself, but I can’t assemble the firmware (Marlin) for this board, since there is no such option in the list of motherboards (chips). There are on G 32 and N 32. If you compile for these boards or some suitable ones, then a black screen comes out after the printer is flashed.
I'm sorry, maybe I'm asking stupid questions, but I started to study this topic relatively recently. I followed all instructions. Used all the latest updates.
Oh yes and more. I flashed your Marlin configuration with my Aquila s 2 settings and everything worked for me.
Thank you for that. I hope someone can help me with my problem with Aquila X2. Thank you all in advance.
nuriknar2016@gmail.com

inspectionsbybob, 2023-03-15:
NurikNar- I think you put this in the wrong area