alexqzd/Marlin #43: Heating Failed

Justus231, 2023-02-19:
### Description

### Steps to Reproduce

1. turns on printer
2. preheat PLA or starts a print 
3. Bed warms up fine to 55 C
4. nozzel try's to warm up to 190C 
5. gets stuck on 29C 
6. gives error [Printer Kill Reason: Heating Failed Restart Required] 

**Expected behavior:** for the printer to warm up and start printing

**Actual behavior:** gives error and needs a restart
#### Additional Information: i use the 1.3.5 version 4x4 N32

[image]

Mik-S-UMS, 2023-02-19:
This could be a problem with your thermistor in the hotend. First thing to do is run a PID tune to make sure the firmware knows how your hotend will heat up. If you have just installed Alex's FW then your old settings may have been overwritten.

Justus231, 2023-02-19:
I just bought a new hot end, beceaus the old one broke (beceaus i was dum)
[image]

Mik-S-UMS, 2023-02-19:
You will definitely need to do a PID tune with a new hotend as its thermal characteristics will be different. Its good practice to redo PID whenever you do work on the hotend.

When replacing it make sure the display shows room temp on both bed and hotend before heating it up. This just makes sure the thermistor is wired up correctly and working. If it shows some abnormal temperature the it is possible there has been damage in the CPU or the thermistor is faulty.

Justus231, 2023-02-20:
when i try to do a PID tune with pronterface, the same error message will pop up

Justus231, 2023-02-24:
Nvm, I have done the dum dum. when I put in the new wires, instead of putting them on the right side of the platform. I put them on the left.