alexqzd/Marlin #46: Voxelab Aquila 1 G32  unable to load firmware

npaisnel, 2023-06-28:
SD card previously formatted as FAT32 4096.

I can happily bounce back and forward between the Original Voxelab Aquila_V1.2.4_20210826 and Aquila_v3.1.2_bltouch_20210827 firmwares no problem.

But any of these I try, nothing .  it just boots to the main screen with no attempt to load the new firmware.

As I say, can happily swap between the two Voxelab firmwares, so not an SD card issue.so ..what can it be
Voxelab Aquila 1 G32

Just trying to get aBL Touch fitted and working ..it does work with the G32 BLT firmware ..but not gee rat when it comes to Z-axis offsets..so wanted to try your firmware.

Any hints appreciate.

Mik-S-UMS, 2023-06-29:
Try renaming the .bin file as if it is the same as previous version installed then it won't accept it.
Also check to see if the .bin file is still there in the firmware folder after trying to install. If it is then it failed.

npaisnel, 2023-06-29:
Hi 
No that did not work either I had already tried that.
I did discover the issue.
yes the card was FAT32 4096 format ..that was fine   
I could write to it on the Apple mac and consistently swap between the OEM 1.2.4 and 3.1.2 factory firmwares..writng them to the card on the Mac   I did this multiple times.

yet when writing the Alexqzd firmware to the same card, with the same adapter, with the same mac  the firmware would to upload to the printer WHATEVER name I gave it .  Yes, it stayed on the card, it was not flashing   After a failure..I could stick an  OEM firmware bin to the SD card and continue to toggle between the two OEM firmwares..but the mac woudl nto write the Alex firmware to the SD card.

Use the same SD card on a Windows laptop..the sea one the card was formatted in, and the Alex firmware flashed just fine.

A screwy Windows / Mac filewrite issue I guess

I have jsut found this slightly later firmware ...hmm...maybe I should try that ...

https://github.com/classicrocker883/Marlin-Alex/releases/tag/v2.0.1

Mik-S-UMS, 2023-06-29:
The problem was the Mac. 
They leave hidden files behind that will cause the installation to fail so is why it is recommended to use a PC to prepare the SDcard. 
There is a way to clean them up with a command in the terminal if you don't have that option, more info [here](https://www.reddit.com/r/VoxelabAquila/comments/odxwdk/psa_before_flashing_custom_firmware_using_macos/).

Alex is no longer working on this firmware any more so you might be interested in [Mriscoc ProUI](https://github.com/classicrocker883/MriscocProUI/releases/tag/2.1.3d-1) which is  similar to Alex's but more up to date with main Marlin and has more features.

npaisnel, 2023-06-29:
thanks for the link to the newer firmware.  will give it a try .

it was funny / odd though how the OEM firmware was not affected buy the Mac hidden files.  I had Finder  (CMD SHIFT . ) set to view hidden files..but I did not check the volume in Terminal.

issue solved nto so , thatnks

Will give the newer link a try ,,,12 hour build at the moment for abetter fan shroud..if I can get the bloody stuff to stick... even using ABL and about 4 hurst's with it this morning ..still nto able to get a satisfactory bed level and Z offset.  beginning to think there is some other issue...

npaisnel, 2023-06-29:
Hmm is see versions listed for GD32 I have G32 ..reckon it is the same ...just had call to go to work ...i fly air ambulance...so not able to check right now, but if  you know?

Mik-S-UMS, 2023-06-29:
The G32 is just an abbreviation of the numbers on the chip which starts GD32... 
Same goes for the H32 which is short for HC32... 

I know little about Macs but from what I have read you will not be able to find those "hidden" files on a Mac as they are part of a legacy file system, that is why only that terminal command will remove them.

I think the reason the stock firmware was not affected is as it is fairly basic it will take up less space in the chip's flash whereas CFW will use up almost every last available byte. I guess the boot loader does a check to see if the new firmware is going to fit but it counts these extra file artefacts too and thinks the file is too big.

npaisnel, 2023-06-29:
G code yes
tried G29  but on the bed levelling sequence I get a temperature runaway warning on about the 4 th level probe activation..so have had to use the M420 S1 to use the stored mesh

G92 E0 ; Reset Extruder
G28 ; Home all axes
G29; Auto Bed Level // can be M420 S1 if don't want to probe before each print
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X0.1 Y20 Z0.3 F5000.0 ; Move to start position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; Draw the first line
G1 X0.4 Y200.0 Z0.3 F5000.0 ; Move to side a little
G1 X0.4 Y20 Z0.3 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z2.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5 Y20 Z0.3 F5000.0 ; Move over to prevent blob squish

Bed level is pretty good according to he "View "mesh option.  Seems to br getting the z offset correct.  
It does not seem consistent.  I can print 20x20 x 0.2 mm flats and they print fine..but then when I print an stl with Cura settings he same on the slicer.. it looks like very squished..but no bed adhesion

that was with a Brim...tryinga Raft now..Raft is looking good...so far...
suppose should have tried that new firmware first...but see how this goes

npaisnel, 2023-06-30:
Got that new firmware running now ProUIwith UBL...still can't get items to stick to the ruddy bed though!   Even 20mm bigger raft is lifting at the corners.  Trying hotter bed temp now to see if that helps..    Thanks for the help

Mik-S-UMS, 2023-06-30:
You probably need to dial in your Z-offset to get good adhesion. 
You should not need to use rafts, I tried one once and the raft printed fine but the print detached straight away. It also wastes too much filament. I now use a wide brim if I think something will not stick.

Might be better to post on the [Aquila reddit](https://www.reddit.com/r/VoxelabAquila/) if you still have problems. 
There is a sticky post that has a lot of tips covering most common problems and if that does not help then post pictures and I'm sure someone will see what is going wrong.

npaisnel, 2023-06-30:
Dialing in Z-offset is about all I have been doing for the last few days ...  printing , doing live adjustments ..watching it look fine until a could of layers in ..then it slips off...tried form squished all the way up to loose then back down to get adhesion.  

Have checked flow rates, E steps/mm  Brim usually works fine, but not this last few times..a raft worked font this time..but yes a lot of wasted filament...print got about 4 hours through..then I relaisd I had printed the wrong item!  

Hey Ho...printing again !

npaisnel, 2023-07-06:
> The problem was the Mac. They leave hidden files behind that will cause the installation to fail so is why it is recommended to use a PC to prepare the SDcard. There is a way to clean them up with a [...]

I have been trying this firmware and it just seems 'buggy'

Getting constant either Low Bed Temp errors or Over temp errors.

It often reports stuff like -14 deg C and then says MINTEMP error   Restart machine .

Go back to original firmware and there is no problem.  I can only guess that it is having some temperature probe sampling errors...not experienced this with the Alex firmware or the OEM..at least can't remember it on Alex firmware

And the whole bed tramming and UBL 'conflict.   Bed tramming shows bed level..eventualy ..  yet the UBL mesh then always shows it high along RH edge...

Maybe a crap BLTouch sensor...or Z-axis travel..Thi smachiein has only one Z axis lead screw and the other side gets 'pulled' up the spot by the fact the LHS is risen g..a mechanically very poor design..  pile of crap..I can't adjust it any better...design limitation...the RH side 'trolley' that rides the Z-axis channel has vertical 'slop...not much maybe a mm or two ..not loose slop ..but definite movement.

Mik-S-UMS, 2023-07-06:
If you have trouble with levelling it could be that your eccentric nuts are too loose so there is sag at one side. Watch [this](https://www.youtube.com/watch?v=Q7Mu05fZNsk) for common levelling problems and how to fix.

For temperature problems try running a [PID tune](https://www.youtube.com/watch?v=vCQWx4MBjjY), you can do it from the menus on the printer with CFW. It may be called MPC on ProUI as it is a different way of doing it.

If it reads -14C then there is something wrong with the thermistor, that is the value you get when it is not connected so there may be an intermittent break in the thermistors wires. 

If it suddenly fixes using another firmware then it could be a bug. see if you can reliably replicate it and open an issue on that repository so it can be fixed. I think there was another release recently so it may be fixed already.

ripptech, 2023-07-06:
"The problem was the Mac."

little tip for the future:

$dot_clean -v /Volumes/<SDCARD>

will get rid of all the cruft your Mac leaves behind and makes it usuable.

npaisnel, 2023-07-07:
Yes, sorry, not good of me, I should have included the command rather than just the link to it.

closed 2023-07-07 by npaisnel

Mik-S-UMS, 2023-07-07:
The previous owner must not have assembled it correctly if they had no luck either. 

I always suggest people that get a second hand printer should disassemble and rebuild it following [this](https://www.youtube.com/watch?v=me8Qrwh907Q) guide to find any hidden problems and make sure everything is correctly aligned and adjusted.

npaisnel, 2023-07-07:
Well, it has worked fine, once you get the first layer to stick

this is a chess set I did with it for him

[image]

[image]

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]